anerdins/nibepi #2: Nibepi with arduino nibegw

veijavi, 2021-01-18:
Hi,

Thanx for the great piece of sotware! I'm using node-red-contrib-nibepi with Arduino nibegw for F1155-12. It seems that it's not working out of the box. I had to do the following modifications to get it working.

**UDP ports**
UDP ports are hard coded in nibepi. 9999 is for incoming data , 10000 is for read commands and 10001 is for write commands. By default nibegw-arduino uses 9999 for sending data and receiving read commands and 10001 for receiving write commands. So to get it working this have to be modified from code in nibegw or nibepi.

**Read and wirte tokens**
With raspberry nibegw version it seems to be possible to use "-i" parameter to send all data to nibepi including read and write tokens. Nibepi seems to require these to send any UDP messages out to nibegw. But arduino nibegw seems to send only the modbus data and there are no options to send read and write tokens to UDP.  Off course the most elegant way would be to pass real received tokens to UDP, but because these tokens are always the same, I hard coded them and modified the code to send them to UDP every time the token is received from RS-485 side.

```
byte readToken[]={0x5C,0x00,0x20,0x69,0x00,0x49};
byte writeToken[]={0x5C,0x00,0x20,0x6B,0x00,0x4B};

    if (token == READ_TOKEN)
    {
      sendUdpPacket(readToken, 6);
      ....
     }
    
    else if (token == WRITE_TOKEN)
    {
      sendUdpPacket(writeToken, 6);
      ...
    }
```

Tobi0610, 2021-01-18:
I can confirm that it is necessary to send these tokens. Otherwise NibePi is not sending any UDP messages/requests as described above.

Thank you!

DenisMir, 2021-11-13:
@veijavi Is this also valid for the current april version of nibegw that the code has to be changed to send the tokens? (https://github.com/openhab/openhab-addons/tree/main/bundles/org.openhab.binding.nibeheatpump/contrib/NibeGW/Arduino/NibeGW)

elupus, 2022-12-10:
The esphome based gw already sends tokens.

elupus, 2022-12-10:
But, default on esphome-nibe is port 9999 for sending, 9999 for reads, 10000 for writes. This matches default on the arduino from openhab, but oddly not the raspberry pi variant.

So you need yo change to 9999, 10000, 10001