anerdins/nibepi #3: non responsive VVM310

esebrani, 2021-01-26:
I was hoping to tweek my NIBE with your nice project but I have run in to trouble. After activating MODBUS I got error code 253, no connection. I deselected it again and also deconnected the Pi. The fault code disappeared but now I got the code 301 , no connection to the heat pump. And now I am stuck. The Pi seems to work as I can see it from my computer. Is there anyway to figure out if the vvm310 is broadcasting signals? I have this feeling that when I turn off the MODBUS my VVM310 also lost connection with the heatpump and now they can't find each other again. And yes I have rebooted everything, updated firmware etc.

anerdins, 2021-02-06:
> I was hoping to tweek my NIBE with your nice project but I have run in to trouble. After activating MODBUS I got error code 253, no connection. I deselected it again and also deconnected the Pi. The [...]

As we talked about in another platform your AA3 card in VVM310 broke during installation of NibePi.
This might be because faulty soldering, bad components, bad installation, etc..
My guess is bad luck. 200 installed nibepi and one of then has now resulted in a broken component in the heatpump.
Statistically this will happend, and Im sorry this happend to you.

esebrani, 2021-02-06:
I am now suspecting my raspberry pi zero as it also destroyed two sd-cards when I was trying to use the rasp in other installations. Sooner or later I will try again with a new hardware...

dennisfisch, 2021-03-24:
My NibePI installation has been running well for about 6 months and I woke up today to the exact same error.
After disconnecting the PI and disabling MODBUS, the VVM320 unit is showing me 301 errors, Heatpump communication fault.

For whatever reason, I suspect the additional load caused by NibePI may eventually destroy the AA3 unit.

dennisfisch, 2021-03-24:
Maybe helpful to others: I shut everything down by flipping the mains switches on all heatpump circuits (internal and external). After resetting and startup, the external heatpump is now showing up again. Seems like it is working properly now.

I am a bit afraid of reconnecting the PI, since damaging any components is probably going to be quite costly and will crush the wife-acceptance-factor into oblivion