anerdins/nibepi #4: VVM320 loses connection to nibeuplink

masc226, 2021-02-04:
My Nibe VVM320 loses the connection to nibeuplink after a couple of days using the nibepi.
I can only reconnect due to restarting the VVM320.
Has anybody else the same issue or is it not possible to use the nibepi and nibeuplink at the same time.

anerdins, 2021-02-06:
> My Nibe VVM320 loses the connection to nibeuplink after a couple of days using the nibepi.
> I can only reconnect due to restarting the VVM320.
> Has anybody else the same issue or is it not possible to use the nibepi and nibeuplink at the same time.

Nibe uplink has generally a lot of issues that resolves with restart of the heatpump. 
So most likely unrelated to nibepi.

SgtSeppel, 2021-02-09:
I can confirm this.
NibeUplink is so unstable that I completely stopped pulling data from it ever since NibePi was stable enough.

Xnygra, 2021-02-17:
I have not start use NibePi, and reson to look into NibePi is because of my VVM310 does the same, looses connection to the cloud. When this happens ill go into Nibe TCP/IP menu and press confirm (without altering anything) and the Nibe crashes and restart itself. So Im about to abandon Nibe Uplink and move over to NibeP.