anerdins/nibepi #8: TypeError [ERR_INVALID_ARG_TYPE]: The "id" argument must be of type string. Received undefined

v1ckdesigns, 2021-10-04:
Hello,
I am trying to run nibepi in a docker container connected with USB to 485 adapter
i have setup /etc/nibepi/config.json to point to the /dev/ttyUSB0 and as soon as i enable modbus on my SMO-40 unit i got an error message :

Waiting for the core to initiate...
About to exit with code: 1
node:internal/validators:119
    throw new ERR_INVALID_ARG_TYPE(name, 'string', value);
    ^

TypeError [ERR_INVALID_ARG_TYPE]: The "id" argument must be of type string. Received undefined
    at new NodeError (node:internal/errors:371:5)
    at validateString (node:internal/validators:119:11)
    at Module.require (node:internal/modules/cjs/loader:998:3)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /usr/app/nibepi/index.js:435:27
    at checkPump (/usr/app/nibepi/index.js:394:9)
    at announcment (/usr/app/nibepi/index.js:433:13)
    at ChildProcess.<anonymous> (/usr/app/nibepi/index.js:270:25)
    at ChildProcess.emit (node:events:390:28)
    at emit (node:internal/child_process:917:12) {
  code: 'ERR_INVALID_ARG_TYPE'
}

any help would be appreciated

v1ckdesigns, 2021-10-04:
i managed to troubleshoot it further, my model is reported as NIBE,BA-SVM,10-200/12,E
i have hardcoded to SMO40 and it seems to work, but i only get the alarm register showing up. 

any steps to discover register values?

szuki, 2021-12-03:
Any progress with it?

tanelvakker, 2022-01-08:
This seems to happen when the model number from the unit does not match any of the https://github.com/anerdins/nibepi/blob/master/lib/models.json 
You should be able to configure this with config.system.pump variable