anerdins/nibepi #9: img size

MichaelPihlblad, 2022-02-08:
img size can be truncated to < 7.5GB, partitons is not bigger than that, lots of empty space. It would make it fit on a 8GB SD card.

`truncate --size=$[(15693823+1)*512] nibepi_1.1.1.img`

```
fdisk -lu ./nibepi_1.1.1.img 
Disk ./nibepi_1.1.1.img: 14,86 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xd6009415

Device              Boot Start      End  Sectors  Size Id Type
./nibepi_1.1.1.img1       8192    98045    89854 43,9M  c W95 FAT32 (LBA)
./nibepi_1.1.1.img2      98304 15693823 15595520  7,4G 83 Linux
```

Btw, successfully running on older Raspberry pi B rev 2 from 2012.
will try if i can make it work with regular USB<->RS485 cable from kjell.com.

How do you feel about adding a wiki and letting anyone contribute with documentation?

Mooomin123, 2022-03-11:
> Btw, successfully running on older Raspberry pi B rev 2 from 2012.

Thank you for confirming this, I might try the same setup. :)

> will try if i can make it work with regular USB<->RS485 cable from kjell.com.

I'm curious, did you manage to try this out?

MichaelPihlblad, 2022-03-11:
I did not manage to get the kjell..com cable working OK. I would advice against using that cable,  with Linux at least.

It is labeled 'cc-usb-rs485-150u'. It is sold for Epever / epsolar communication. It has a Maxlinear (previously Exar) circuit.

The universal kernel driver 'cdc_acm' in this Raspbian image I only got reading to work, no writing. 

Same when trying to update kernel and config to include the specific xr_serial driver for this chip. 

Tried compiling the driver from Maxlinear download. Got it working for a short while, both read and write.  Suddenly stopped working completely, neither read nor write.   I did not manage to get it working again. So I gave up and bought another standard usb to rs485 cable from Elfa distrelec in Sweden which worked fine from start, true plug and play.

So i recommend to stay away from Maxlinear RS485 cable if using Linux. Any other standard FTDI cable should work just fine.