anerdins/nibepi #10: Compatible with Pi Zero 2 W

ppent, 2022-04-09:
It's difficult to get a Pi Zero W at the moment. Will this project also work with the newer version 1 W or is the power consumption too high with that model?
Thank you!

Pete1979, 2022-04-17:
Its about impossible to get a Zero W, Compatibility with Zero 2 W is kinda needed.

anerdins, 2022-04-18:
The newest image should be compatible with raspberry zero 2. No one has confirmed it though.

anerdins, 2022-04-18:
No measurements has been made. Try research the power consumption of the new board.

steefan85, 2022-06-07:
I am using a Pi Zero 2 W with a F730 since January 2022 without any problems; works great!
No changes necessary wotks out of the box.

ppent, 2022-06-15:
> I am using a Pi Zero 2 W with a F730 since January 2022 without any problems; works great! No changes necessary works out of the box.

Just to make sure: You don't use any external power supply, correct?

steefan85, 2022-06-15:
> > I am using a Pi Zero 2 W with a F730 since January 2022 without any problems; works great! No changes necessary works out of the box.
> 
> 
> 
> Just to make sure: You don't use any external power supply, correct?

Yes, that's correct. I'm using the Wide Input Shim directly connected to the 12V Connection on the F730 Mainboard.

MoYzOne1, 2022-07-15:
> Yes, that's correct. I'm using the Wide Input Shim directly connected to the 12V Connection on the F730 Mainboard.

I have a Pi Zero 2 W too - revision 2.13 but the image doesn't work out of the box. The WiFi interface is missing. With a normal Buster image it works. Any ideas ? Maybe a missing driver or something. Is there a way to combine the NibePi image with a up to date buster ?

Greetz MoYz

steefan85, 2022-07-15:
> I have a Pi Zero 2 W too - revision 2.13 but the image doesn't work out of the box. The WiFi interface is missing. With a normal Buster image it works. Any ideas ? Maybe a missing driver or [...]

Hi,

am running on Debian Stretch, so maybe my image-file [nibepi_1.1.rar](http://anerdins.se/NibePi/nibepi_1.1.rar) was not the same as yours.

The newest Version [nibepi_1.1.1.rar](https://github.com/anerdins/nibepi/blob/master/README.en.md) was built for Debian Bullseye

MoYzOne1, 2022-07-15:
> Hi,
>
> am running on Debian Stretch, so maybe my image-file [nibepi_1.1.rar](http://anerdins.se/NibePi/nibepi_1.1.rar) was not the same as yours.
>
> The newest Version [...]

I am indeed using the 1.1.1 image. But it should work too. Very strange. Thank you for your reply. I will try the Stretch image. If WiFi is running there, maybe I can upgrade to Bullseye from there.

MoYzOne1, 2022-07-16:
> I am indeed using the 1.1.1 image. But it should work too. Very strange. Thank you for your reply. I will try the Stretch image. If WiFi is running there, maybe I can upgrade to Bullseye from there.

Using the 1.1 image the Pi doesn't even boot (Rainbow Screen) ... original Rasbian Lite images boot with WiFi without any issues. I dont't understand this :(. I thought: hey: nice to have an image. Just write it to SD and your done. Seems like no NibePi for me.

vytautasmiel, 2022-07-22:
To launch NibePi on Raspberry Pi Zero 2 W and get it connected via wifi I had to add firmware files brcmfmac43436-sdio.bin, brcmfmac43436-sdio.txt to /lib/firmware/brcm/ folder.

MoYzOne1, 2022-07-22:
Thank you for your answer - Meanwhile I managed to make a clean install of Node Red importing the NibePi flows.