anerdins/nibepi #11: NibePi docker error

frankvandenhurk, 2022-05-10:
When I try to start docker with the default docker-compose i get:

```
root@debian1:/home/nibepi# docker-compose up
Starting nibepi ... done
Attaching to nibepi
nibepi    | Waiting for the core to initiate...
nibepi    | Error in the core, shuting it down.
nibepi    | About to exit with code: 0
nibepi exited with code 0
```

Anyone a clue where to start looking for what could be wrong?

frankvandenhurk, 2022-05-10:
With debug on:

```
nibepi@debian1:~/config$ docker-compose --log-level DEBUG up
Using configuration files: ./../docker-compose.yml
Trying paths: ['/home/nibepi/.docker/config.json', '/home/nibepi/.dockercfg']
No config file found
Trying paths: ['/home/nibepi/.docker/config.json', '/home/nibepi/.dockercfg']
No config file found
http://localhost:None "GET /v1.25/info HTTP/1.1" 200 None
http://localhost:None "GET /v1.25/containers/json?limit=-1&all=0&size=0&trunc_cmd=0&filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 3
http://localhost:None "GET /v1.25/containers/json?limit=-1&all=0&size=0&trunc_cmd=0&filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 3
http://localhost:None "GET /v1.25/containers/json?limit=-1&all=1&size=0&trunc_cmd=0&filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 1490
http://localhost:None "GET /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/json HTTP/1.1" 200 None
http://localhost:None "GET /v1.25/containers/json?limit=-1&all=1&size=0&trunc_cmd=0&filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.service%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 1490
http://localhost:None "GET /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/json HTTP/1.1" 200 None
http://localhost:None "GET /v1.25/images/anerdins/nibepi-base/json HTTP/1.1" 200 None
http://localhost:None "GET /v1.25/containers/json?limit=-1&all=1&size=0&trunc_cmd=0&filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.service%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 1490
http://localhost:None "GET /v1.25/images/anerdins/nibepi-base/json HTTP/1.1" 200 None
http://localhost:None "GET /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/json HTTP/1.1" 200 None
Pending: {<Service: nibepi>}
Starting producer thread for <Service: nibepi>
Starting nibepi ...
Pending: {<Container: nibepi (a6a0c6)>}
Starting producer thread for <Container: nibepi (a6a0c6)>
http://localhost:None "POST /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/attach?logs=0&stdout=1&stderr=1&stream=1 HTTP/1.1" 101 0
http://localhost:None "GET /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/json HTTP/1.1" 200 None
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
Pending: set()
http://localhost:None "POST /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/start HTTP/1.1" 204 0
Starting nibepi ... done
Pending: set()
Finished processing: <Service: nibepi>
Pending: set()
Attaching to nibepi
nibepi    | Waiting for the core to initiate...
http://localhost:None "GET /v1.25/events?filters=%7B%22label%22%3A+%5B%22com.docker.compose.project%3Dnibepi%22%2C+%22com.docker.compose.oneoff%3DFalse%22%5D%7D HTTP/1.1" 200 None
nibepi    | Error in the core, shuting it down.
nibepi    | About to exit with code: 0
http://localhost:None "GET /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/json HTTP/1.1" 200 None
http://localhost:None "POST /v1.25/containers/a6a0c62d3d47f44250996f682819c2f757eb6b6306ddb3a90d8423c83fd1dbe2/wait HTTP/1.1" 200 30
nibepi exited with code 0

```

MichaelPihlblad, 2022-05-21:
I have the same issue.

I concluded it is due to non existing serial device communication that causes `server.js` to exit.

I never bothered to proceed and get serial communication to work in Docker because I honestly don't think it would add much value for me to use this docker image. I guess it is only useful when developing the Node.js stuff? 

I guess you can either hook your docker host up to your heatpump or create a NIBE simulator or something if you want to get it working.

i.e. try setting ethernet device instead of serial in the config.json file used by docker (configured in `docker-compose.yml` under `volumes:` by which host folder containing the file is mounted in container folder: `/etc/nibepi`)
`config.json`:
```diff
"connection": {
-  "enable": "serial",
+ "enable": "nibegw",
```

when i did that i instead got:
```bash
$ docker-compose up
Starting nibepi ... done
Attaching to nibepi
nibepi    | Waiting for the core to initiate...
nibepi    | NibeGW client is listening on /dev/ttyAMA0:9999
```

frankvandenhurk, 2022-08-16:
Finally I had time to take another look. Created a Debian VM wit node12 and nodered installed and installed the nodered contrib. Exported all flows from the running Raspberry with nibepi and imported them in the VM. Same error. 

Then I realised that the user node-red was running as, didn't have enough rights to access /dev/ttyUSB0. Added the user to the group dialout (sudo usermod -a -G dialout $USER) and it works like a charm! Sometimes you can feel really stupid not thinking about the obvious.

closed 2022-08-16 by frankvandenhurk