anerdins/nibepi #15: Alternative hardware

gustafnilklint, 2022-08-06:
It's really hard to find the rs485 hat that you refer to in the documentation. It seems to be sold out everywhere.
Is it possible to use a different module for the modbus? Can you recommend an alternative?

frankvandenhurk, 2022-08-16:
I use the MAX485 USB module for RS485. It's the black USB stick with the two green screw connectors that you can find everywhere online for about 4 euro/dollar. Connect it to a regular Raspberry Pi (with USB connectors) at you're good to go.

(found out that for RPI4 the kernel was missing in the image, so I had to transplant that from a working RPi image)

hado010, 2022-08-18:
I don't quite understand this comment: "found out that for RPI4 the kernel was missing in the image, so I had to transplant that from a working RPi image".
It is not clear to me whether or not I have to do something special/extra, if I want to build a system with the above mentioned hardware?

frankvandenhurk, 2022-08-21:
Id I recall it correctly, I had to download and image a normal rapberry 4 image from https://www.raspberrypi.com/software/ and copy some files from the boot partition to the boot partition of NebiPi in orde to get it work, but after that it works like a charm.

closed 2022-09-02 by gustafnilklint