anerdins/nibepi #20: Question: energy consumption

DiiJAYA, 2022-09-27:
Hi,

Looking to maybe connect my F750 with nibepi, I'm interested in power/energy consumption.
Does this integration provides current energy consumption in kW and total energy consumption in kWh, if so in what parameters will I find these values?

Thanks in Advance..

AcEIoXx, 2023-02-06:
As far as I know no models from nibe measure this on their own.
There are models that measure whole house, but only to prevent blowing the main fuses.

But I could be wrong.

AnonymousRetard, 2023-05-28:
I think mine has this (F750) but I'm not 100% sure I have done the right calculations to get the right numbers. I have not tried to read these values from Modbus yet but for now I am using an integration with NIBE uplink through home assistant.

I imagine Modbus would have the same numbers though and perhaps more.

Already from the start I have parameters called:
- "Compressor power used, total [kWh]" (sensor.nibe_149100_43144)
- "Compressor power used, hot water [kWh]" (sensor.nibe_149100_43305)

I think those are pretty self-explanatory and they seem to give me reasonable values. On top of these I calculated my own parameter which I call "Compressor power used, radiators [kWh]" which is just the total minus the hot water one.

It seems like those two don't always update at exactly the same time which means that unfortunately sometimes it looks like the power used for the radiators regresses about ~-0.6 kWh or so but I have not seen much larger numbers than that.

Then there's a parameter without much explanation available called "time factor (sensor.nibe_149100_43081)", but I *think* you can convert that one to kWh used for electrical addition ("tillsats" in Swedish) by multiplying it by "set max electrical add." (sensor.nibe_149100_47212). This is the currently set max power limit of the electrical addition. That's what I do at least and the numbers I have been getting out from this seems reasonable...

I'm not sure if the sensor.nibe_XXXXXX_XXXXX numbers are helpful to you but I imagine they could have corresponding Modbus registers.