anerdins/nibepi #21: Nibe F1240

nuiler, 2022-09-29:
I have a Nibe heat pump F1240 can I also integrate it? 

Does this also work with a USB-RS485 adapter?

anerdins, 2022-09-29:
It is too old unfortunally. Wrong series

nuiler, 2022-10-05:
oo bad. Nevertheless, do they perhaps have the settings for Modbus RTU(baud,etc.)? Would like to connect them to an Ethernet Modbus converter. I would like to try to connect them also via Node-Red

superwolfi82, 2024-05-31:
@nuiler  did you make any progress on this, i also got a F1240, i need to connect

MichaelPihlblad, 2024-06-01:
I would recommend to check out ESP home Nibe integration.
 
https://github.com/elupus/esphome-nibe

https://www.home-assistant.io/integrations/nibe_heatpump/

nuiler, 2024-06-01:
> @nuiler did you make any progress on this, i also got a F1240, i need to connect

I didn't pursue it any further because I bought the module from husdata.se.