anerdins/nibepi #25: No data from smo40? No id?

DanielAscher, 2022-10-31:
Hi,

I tried nibepi (although not the image itself, but node Red with nibepi node Red.

I can sconnect it to the pump. The pump stays Green. I can see some values in the logs. But I cant see any in node Red.

I also got an error for "id" undefined so I had to Hardcode Model =" SMO40" in Index.js... That made it kinda work (no modbus error on the pump". But as said... No values are displayed.

Also, how can I have home Assistent connect to nibepi?

Also, and I think that's the problem, I have a nibe split. "BA-SVM 10-200" it says on the pump. In uplink I can see it should be an "SMO40"?

Any help is much appriciated.

DanielAscher, 2022-11-02:
No idea?

AcEIoXx, 2023-02-06:
Most likely same issue as with issue 8
BA-SVM is unknown as a model.

You could try coping the SMO40 (as that is the base for the BA-SVM) to a new model and start from there.

tomeczko, 2024-02-22:
> No idea?

Hi Daniel! Did you eventually suceed with connecting nibepi to BA-SVM 10-200? I have the same internal unit and also am strugging with marrying it with nibepi :) I have checked the logs reported by BA-SVM on serial lines and there is explicitly log: "BA SVM 10-200" (with spaces etc). I'm not sure if this is the reason why nibepi fails to detect the unit... :thinking: