anerdins/nibepi #28: Another RS485 HAT

andreasropel, 2023-02-27:
Is NibePi compatible with one, and only one type of RS485 HAT? Or could this be used: https://www.amazon.se/IBest-RS485-l%C3%A5ngdistanskommunikation-CAN-Module/dp/B07PQJMFVP/ref=sr_1_6?keywords=rs485%2Braspberry%2Bpi&qid=1677514526&sprefix=rs485%2Bras%2Caps%2C95&sr=8-6&th=1

MichaelPihlblad, 2023-02-27:
Any RS-485 UART should work fine I assume. I use an external USB connected device to a Raspberry v3.  I guess the only important detail is that the HAT is compatible with your raspberry in respect of pinout etc.

hoegge, 2023-09-03:
But what is the RS-485 HAT needed for? I thought the NibePi  communicated with heat pump controller over serial and then to the outside world over WiFi?